Cozmonat/HWSensors #269: Future Ideas

audioslave74, 2015-02-01:
I have 2 ideas for you.
First that you would add fan speed profiles, for example: silent-normal-exreme, or or the ability to add multiple custom profiles.
Second to add settings for remoting control the fan speed profiles with bluetooth remote controls.

perfaram, 2015-04-21:
See my comment in #275 
EDIT : Here it is
[...] I had an idea to help fixing all those GPU identification problems : 
http://pci-ids.ucw.cz/pci.ids.
It's a nice (and *regularly updated* since 2012 at least) list of PCI vendors/devices. 
For example, it tells you that there has been 10 GT730 versions (luckily they all have their *own* hex device ID); one of them being `1287  GK208 [GeForce GT 730]`.
It could help @kozlek having a smaller code base, as we would know the chipset (GK208 here) without having those hardcoded `switch (device->chipset)`.
To help you parsing this file : 
* https://github.com/dcdillon/cpuaff/blob/bab8df074aee6b19790f30aac15aa5892c340db0/include/cpuaff/pci_name_resolver.hpp
* See the `readIDs:` selector in this ObjC file : https://github.com/walkman8196/DPCIManager/blob/master/DPCIManager/PCI.m

ilanbiala, 2016-02-20:
Profiles would help a lot.