Cozmonat/HWSensors #270: HWSensors Fan's name issue

troydm, 2015-02-20:
I have latest HWSensors installed running HWMonitor 6.14.1364
My hardware is Macbook Pro 13 retina mid-2014 MGX72LL/A

Some strange &sy characters are appended to the name of my Macbook's right side fan
[image]

swmuron, 2015-02-22:
Also getting this issue on MBP 13" Retina Late-2013 / MacBookPro11,1.

Sometimes multiple buggy "Right Side [garbage]"  fan names appear, despite the model only having one fan.

troydm, 2015-07-07:
after installing latest update this has been fixed, thx a lot guys!

troydm, 2015-07-16:
unfortunately in latest version it's not completely fixed, there are no artifacts in the name however if I add fan sensor output to panel after restart it's not remaining

fbnk, 2015-08-02:
I have the same issue with my MBPr Mid 2014, and would love to get an update for this. Thx!

avidrissman, 2016-02-25:
I see this on my MBP10,2 with v 6.18.1394.