Cozmonat/HWSensors #275: GeForceSensors (pci1):[Fatal] unknown Kepler chipset 0x106,  [Fatal] unsupported chipset, 0xb06070b1

ericj, 2015-04-19:
Hi, @kozlek  

I am using FakeSMC v6.13.1363 with a GT 730 kepler GK208 video card, and HWSensors seems not support this. 
I think it might similar with GT630 v2 GK107? 
Is it possible to be supported? 

Thank you!

perfaram, 2015-04-19:
It doesn't really help you, but I had an idea to help fixing all those GPU identification problems : 
http://pci-ids.ucw.cz/pci.ids.
It's a nice (and regularly updated since 2012 at least) list of PCI vendors/devices. 
For example, it tells you that there has been 10 GT730 versions (luckily they all have their *own* hex device ID); one of them being `1287  GK208 [GeForce GT 730]` (so we know the chipset without having a full list hardcoded).

MysticalOS, 2015-06-08:
http://sourceforge.net/projects/hwsensors3.hwsensors.p/

Looks like a fork that's currently better maintained at moment. This already supports maxwell, has for several months. Supports GK204 and GK208