Cozmonat/HWSensors #283: How add Fan Control with HWMonitor app

florentino97425, 2015-05-08:
Hi, 
I can control my fan with DSDT Device(SMCD),
but I need to know, how can I add control fan support for hwmonitor app. 
Can someone help me, please.
thanks.

Mother Board HM77 mobile express (laptop)
\_SB.PCI0.LPCB.EC0.TACH = Fan Speed
\_TZ.RTMP = CPU Temp
\_SB.PCI0.LPCB.EC0.SFNV = Fan Control Value

[image]

perfaram, 2015-05-09:
Take a look at : https://github.com/RehabMan/Laptop-DSDT-Patch/blob/master/misc/control_AcerFan.txt

perfaram, 2015-05-09:
http://www.hackintoshosx.com/files/file/4515-dsdt-acer-aspire-v3-771-patched/ Seems good

florentino97425, 2015-05-09:
Both don't help, thanks
the second link, talk about control with Device(SMCD), I can already do it.
But my problem is in hwmonitor app I the control fan menu isn't avaible.

perfaram, 2015-05-09:
AH, ok, sorry. Then, take a look at #239. It's a known fact that HWMonitor disables the fan control whenever a Fintek chip is detected... Is it your case ?Anyway, in #239, @kozlek says "There is no fan control support for Fintek chipsets for now. May be in future I'll make some research but no promises."
Date: Sat, 9 May 2015 00:04:44 -0700

florentino97425, 2015-05-09:
No it's not my case I have product name: VA70_HC (bios name)
my chip is intel hm77 express chipset.
Thanks.

florentino97425, 2015-05-13:
8086, 1E57, 1025, 0685, Intel Corporation, HM77 Express Chipset LPC Controller

perfaram, 2015-05-13:
I was talking about the fan/sensors controller chip, not the system chipset...

perfaram, 2015-05-13:
BTW : 8086 = Intel, 1025 = Acer

florentino97425, 2015-05-15:
Fan/sensors controller chip not avaible in DCPIManager.
Can't find in acer site too.