Cozmonat/HWSensors #290: HWSensor Crash right after start

madman9999, 2015-07-06:
Process:               HWMonitor [1123]
Path:                  /Applications/HWMonitor.app/Contents/MacOS/HWMonitor
Identifier:            org.hwsensors.HWMonitor
Version:               6.17.1386 (1386)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           HWMonitor [1123]
User ID:               503

Date/Time:             2015-07-06 20:11:04.247 +0200
OS Version:            Mac OS X 10.10.4 (14E46)
Report Version:        11
Anonymous UUID:        27E093AF-56CF-9A17-B3B6-B3EA29DE6277

Time Awake Since Boot: 310 seconds

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'this method is supposed to only be invoked on top level items'
terminating with uncaught exception of type NSException
abort() called

madman9999, 2015-07-06:
All older version were ok, current running version is: 6.15.1370

mackonsti, 2015-07-10:
I would like to confirm this crashing. On Mavericks 10.9.5 running latest v6.17.1386 on both my machines, upon starting all seems "stable" but clicking on the data (of the toolbar) or right-clicking on toolbar area, causes the app to crash. Log:

10/7/15 18:33:33,622 HWMonitor[259]: *** Assertion failure in +[NSView _findFirstKeyViewInDirection:forKeyLoopGroupingView:], /SourceCache/AppKit/AppKit-1265.21/AppKit.subproj/NSView.m:12824
10/7/15 18:33:33,644 HWMonitor[259]: An uncaught exception was raised
10/7/15 18:33:33,644 HWMonitor[259]: this method is supposed to only be invoked on top level items
10/7/15 18:33:33,645 HWMonitor[259]: *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'this method is supposed to only be invoked on top level items'
10/7/15 18:33:34,563 com.apple.launchd.peruser.501[164]: (org.hwsensors.HWMonitor.15168[259]) Job appears to have crashed: Abort trap: 6

neuralnetworkz, 2015-07-17:
It's the same for me. last working version is 6.15.1370 on 10.9.5
all newer versions segfault