Cozmonat/HWSensors #291: Un-Reproducible Kernel Panic(s)

winsock, 2015-07-11:
I was mulling about filling this issue since I'm on DP3 however I would rather have people send me too much diagnostic reports for my programs than too little so here is some Kernel panic logs that happened twice.
One that confused me was yesterday I powered up my machine and boom immediate KP with FakeSMC at the top of the backtrace. Rebooted with no changes or attempts to mitigate and it booted just fine. I booted up again today just fine with no issues. I also attached another KP that happened two days ago that I don't recall but was in my DiagnosticReports folder.

[Kernel_2015-07-10-120716_iMac.panic](https://s3.amazonaws.com/randomjunks/Kernel_2015-07-10-120716_iMac.panic)
[Kernel_2015-07-09-143842_iMac.panic](https://s3.amazonaws.com/randomjunks/Kernel_2015-07-09-143842_iMac.panic)

Note: My server will archive these files to tape drive storage after 1 month.
(Why doesn't Github allow uploading of text files -.-)

winsock, 2015-07-11:
Heh, right after I said it was fine I got another. I installed the Supplemental update to 10.11 DP3 that fixes some issues with 32-bit apps and on restart it crashed and took 4 restarts till it booted successfully. Each time a different KP but all have FakeSMC in the backtrace. For some reason only one KP was logged though:
```
Anonymous UUID:       360F0ED9-3D72-6ECC-2E66-C2F2C451A9E8

Sat Jul 11 07:33:30 2015

*** Panic Report ***
panic(cpu 4 caller 0xffffff80003d7e1a): Kernel trap at 0xffffff800083efbd, type 13=general protection, registers:
CR0: 0x0000000080010033, CR2: 0xffffff80f38a6000, CR3: 0x0000000015812000, CR4: 0x00000000001626e0
RAX: 0xffffff80241f20e0, RBX: 0x0000000000000011, RCX: 0x0000000000010002, RDX: 0x0000000000000000
RSP: 0xffffff811393bbe0, RBP: 0xffffff811393bc20, RSI: 0xffffff8000b24078, RDI: 0xffffff80241fd440
R8:  0xffffff80241fd740, R9:  0x0000000000000001, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0xffffff80241fd740, R13: 0xffffff80241fd380, R14: 0x59fcc5d259fcc5c9, R15: 0xffffff8000b24078
RFL: 0x0000000000010206, RIP: 0xffffff800083efbd, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff80f38a6000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff81138fddf0 : 0xffffff80002e74d7 
0xffffff81138fde70 : 0xffffff80003d7e1a 
0xffffff81138fe050 : 0xffffff80003f5553 
0xffffff81138fe070 : 0xffffff800083efbd 
0xffffff811393bc20 : 0xffffff7f830bbf37 
0xffffff811393bc60 : 0xffffff7f830bc008 
0xffffff811393bcd0 : 0xffffff7f830be24f 
0xffffff811393bd00 : 0xffffff7f830bdee5 
0xffffff811393bd50 : 0xffffff7f830d4cf3 
0xffffff811393bd70 : 0xffffff7f830d5c35 
0xffffff811393be80 : 0xffffff8000895add 
0xffffff811393bec0 : 0xffffff80008955e6 
0xffffff811393bf30 : 0xffffff800089029a 
0xffffff811393bf70 : 0xffffff800089627c 
0xffffff811393bfb0 : 0xffffff80003d2857 
      Kernel Extensions in backtrace:
         org.netkas.driver.FakeSMC(1372.0)[D68576F1-B949-3130-9A6A-7B460619B92D]@0xffffff7f830ba000->0xffffff7f830d2fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[CBAE26D8-0ACB-3C1F-8347-FDCA67EC40B3]@0xffffff7f81239000
         org.hwsensors.driver.CPUSensors(1372.0)[6463CC48-8DEE-3060-8246-CFDE56B4ED3E]@0xffffff7f830d3000->0xffffff7f830d9fff
            dependency: org.netkas.driver.FakeSMC(1372)[D68576F1-B949-3130-9A6A-7B460619B92D]@0xffffff7f830ba000

BSD process name corresponding to current thread: kernel_task
Boot args: npci=0x3000 rootless=0 slide=0 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 15.0.0: Sun Jun 28 00:25:56 PDT 2015; root:xnu-3247.1.36~7/RELEASE_X86_64
Kernel UUID: E6DB13C9-258E-310A-A791-6F9E324C3726
__HIB  text base: 0xffffff8000100000
System model name: iMac13,2 (Mac-FC02E91DDD3FA6A4)

System uptime in nanoseconds: 4111480279595
last loaded kext at 4111187625797: com.apple.BootCache	37 (addr 0xffffff7f81999000, size 49152)
loaded kexts:
org.hwsensors.driver.LPCSensors	1372
org.hwsensors.driver.GPUSensors	1372
com.insanelymac.IntelMausiEthernet	2.0.0
org.voodoo.driver.VoodooTSCSync	1.1
org.hwsensors.driver.CPUSensors	1372
org.netkas.driver.FakeSMC	1372
com.apple.BootCache	37
com.apple.driver.AppleIntelCPUPowerManagementClient	218.0.0
com.apple.iokit.IOAHCIBlockStorage	2.8.0
com.apple.driver.usb.AppleUSBEHCIPCI	1.0.1
com.apple.driver.AppleAHCIPort	3.1.5
com.apple.driver.AppleACPIButtons	4.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	163
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleIntelCPUPowerManagement	218.0.0
com.apple.driver.AppleXsanScheme	3
com.apple.driver.AppleUSBMergeNub	900.4.1
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.driver.usb.AppleUSBEHCI	1.0.1
com.apple.iokit.IOAHCIFamily	2.8.0
com.apple.iokit.IONetworkingFamily	3.2
com.apple.driver.AppleEFINVRAM	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.driver.AppleEFIRuntime	2.0
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	414
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.iokit.IOUSBHostFamily	1.0.1
com.apple.driver.AppleUSBHostMergeProperties	1.0
com.apple.driver.AppleACPIPlatform	4.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
```

banzu, 2015-07-12:
Getting the same issue here on OS X 10.11 DP3 SU1. Updated HWSensors last night, probably around the time that the Supplemental Update was installed. Seemingly random as to when I can get past the KP.

````
Anonymous UUID:       6042CAF5-254B-0839-646E-74B2A559670F

Sun Jul 12 09:28:24 2015

*** Panic Report ***
panic(cpu 2 caller 0xffffff80003d7e1a): Kernel trap at 0xffffff800083e404, type 13=general protection, registers:
CR0: 0x0000000080010033, CR2: 0xffffff80e62a5000, CR3: 0x00000000089ee000, CR4: 0x00000000001626e0
RAX: 0xffffff8016888300, RBX: 0xffffff8016891a80, RCX: 0xffffff8016888500, RDX: 0x0000000000000001
RSP: 0xffffff810633bb90, RBP: 0xffffff810633bbc0, RSI: 0xffffff8000b24078, RDI: 0xffffff8016888400
R8:  0xffffff8015a03000, R9:  0x0000000000000200, R10: 0x0000000000000000, R11: 0xffffff8016890d40
R12: 0xffffff8016891a01, R13: 0x0000000000000000, R14: 0xffffff8016888800, R15: 0x0000000000000016
RFL: 0x0000000000010286, RIP: 0xffffff800083e404, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff80e62a5000, Error code: 0x0000000000000000, Fault CPU: 0x2, PL: 0

Backtrace (CPU 2), Frame : Return Address
0xffffff81062f4df0 : 0xffffff80002e74d7 
0xffffff81062f4e70 : 0xffffff80003d7e1a 
0xffffff81062f5050 : 0xffffff80003f5553 
0xffffff81062f5070 : 0xffffff800083e404 
0xffffff810633bbc0 : 0xffffff800083e451 
0xffffff810633bbd0 : 0xffffff800083ef78 
0xffffff810633bc20 : 0xffffff7f81ea6f37 
0xffffff810633bc60 : 0xffffff7f81ea7008 
0xffffff810633bcd0 : 0xffffff7f81ea924f 
0xffffff810633bd00 : 0xffffff7f81ea8ee5 
0xffffff810633bd50 : 0xffffff7f81f01cf3 
0xffffff810633bd70 : 0xffffff7f81f0316f 
0xffffff810633be80 : 0xffffff8000895add 
0xffffff810633bec0 : 0xffffff80008955e6 
0xffffff810633bf30 : 0xffffff800089029a 
0xffffff810633bf70 : 0xffffff800089627c 
0xffffff810633bfb0 : 0xffffff80003d2857 
      Kernel Extensions in backtrace:
         org.netkas.driver.FakeSMC(1394.0)[2DBA2F88-A115-386A-85D0-F88021A4D2E3]@0xffffff7f81ea5000->0xffffff7f81ec1fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[CBAE26D8-0ACB-3C1F-8347-FDCA67EC40B3]@0xffffff7f8176e000
         org.hwsensors.driver.CPUSensors(1394.0)[2A0C01CE-A1E3-399C-BB84-00148F49D473]@0xffffff7f81f00000->0xffffff7f81f07fff
            dependency: org.netkas.driver.FakeSMC(1394)[2DBA2F88-A115-386A-85D0-F88021A4D2E3]@0xffffff7f81ea5000

BSD process name corresponding to current thread: kernel_task
Boot args: -gux_defer_usb2 kext-dev-mode=1 rootless=0 nvda_drv=1  slide=0 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 15.0.0: Sun Jun 28 00:25:56 PDT 2015; root:xnu-3247.1.36~7/RELEASE_X86_64
Kernel UUID: E6DB13C9-258E-310A-A791-6F9E324C3726
__HIB  text base: 0xffffff8000100000
System model name: iMac13,2 (Mac-FC02E91DDD3FA6A4)

System uptime in nanoseconds: 7308130602
last loaded kext at 6547869218: com.apple.BootCache	37 (addr 0xffffff7f82049000, size 49152)
loaded kexts:
com.insanelymac.AtherosL1cEthernet	1.2.3
org.hwsensors.driver.GPUSensors	1394
org.hwsensors.driver.LPCSensors	1394
org.hwsensors.driver.ACPISensors	1394
org.hwsensors.driver.CPUSensors	1394
org.netkas.driver.FakeSMC	1394
com.apple.BootCache	37
com.apple.driver.AppleIntelCPUPowerManagementClient	218.0.0
com.apple.iokit.SCSITaskUserClient	3.7.7
com.apple.iokit.IOAHCIBlockStorage	2.8.0
com.apple.driver.AppleAHCIPort	3.1.5
com.apple.driver.usb.AppleUSBEHCIPCI	1.0.1
com.apple.driver.AppleACPIButtons	4.0
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	163
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleIntelCPUPowerManagement	218.0.0
com.apple.driver.AppleHIDKeyboard	180
com.apple.driver.usb.IOUSBHostHIDDevice	1.0.1
com.apple.driver.usb.AppleUSBHostCompositeDevice	1.0.1
com.apple.driver.usb.AppleUSBHub	1.0.1
com.apple.driver.AppleEFINVRAM	2.0
com.apple.driver.AppleUSBMergeNub	900.4.1
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.iokit.IOSCSIMultimediaCommandsDevice	3.7.7
com.apple.iokit.IOBDStorageFamily	1.8
com.apple.iokit.IODVDStorageFamily	1.8
com.apple.iokit.IOCDStorageFamily	1.8
com.apple.driver.AppleXsanScheme	3
com.apple.iokit.IOAHCISerialATAPI	2.6.2
com.apple.iokit.IOSCSIArchitectureModelFamily	3.7.7
com.apple.iokit.IONetworkingFamily	3.2
com.apple.iokit.IOAHCIFamily	2.8.0
com.apple.driver.usb.AppleUSBEHCI	1.0.1
com.apple.driver.usb.AppleUSBXHCIPCI	1.0.1
com.apple.driver.usb.AppleUSBXHCI	1.0.1
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.driver.AppleEFIRuntime	2.0
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	414
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.iokit.IOUSBHostFamily	1.0.1
com.apple.driver.AppleUSBHostMergeProperties	1.0
com.apple.driver.AppleACPIPlatform	4.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
````

winsock, 2015-07-13:
Here's another log.
```
Anonymous UUID:       360F0ED9-3D72-6ECC-2E66-C2F2C451A9E8

Mon Jul 13 06:46:07 2015

*** Panic Report ***
panic(cpu 4 caller 0xffffff80003d7e1a): Kernel trap at 0xffffff800083e3fa, type 13=general protection, registers:
CR0: 0x0000000080010033, CR2: 0xffffff80f37ae000, CR3: 0x0000000015812000, CR4: 0x00000000001626e0
RAX: 0x0000000000000013, RBX: 0xffffff80240f6d80, RCX: 0xffffff80240e8140, RDX: 0x0000000000000001
RSP: 0xffffff8113843b90, RBP: 0xffffff8113843bc0, RSI: 0xffffff8000b24078, RDI: 0xc0ffeec1ba8be710
R8:  0xffffff8022f0c000, R9:  0x0000000000000002, R10: 0xffffff80233124b0, R11: 0xffffff80240f08d0
R12: 0xffffff80240f6e01, R13: 0x0000000000000013, R14: 0xffffff80240e80a0, R15: 0x0000000000000014
RFL: 0x0000000000010282, RIP: 0xffffff800083e3fa, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff80f37ae000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff8113805df0 : 0xffffff80002e74d7 
0xffffff8113805e70 : 0xffffff80003d7e1a 
0xffffff8113806050 : 0xffffff80003f5553 
0xffffff8113806070 : 0xffffff800083e3fa 
0xffffff8113843bc0 : 0xffffff800083e451 
0xffffff8113843bd0 : 0xffffff800083ef78 
0xffffff8113843c20 : 0xffffff7f830bbf37 
0xffffff8113843c60 : 0xffffff7f830bc008 
0xffffff8113843cd0 : 0xffffff7f830be24f 
0xffffff8113843d00 : 0xffffff7f830bdee5 
0xffffff8113843d50 : 0xffffff7f830d4cf3 
0xffffff8113843d70 : 0xffffff7f830d5cc2 
0xffffff8113843e80 : 0xffffff8000895add 
0xffffff8113843ec0 : 0xffffff80008955e6 
0xffffff8113843f30 : 0xffffff800089029a 
0xffffff8113843f70 : 0xffffff800089627c 
0xffffff8113843fb0 : 0xffffff80003d2857 
      Kernel Extensions in backtrace:
         org.netkas.driver.FakeSMC(1372.0)[D68576F1-B949-3130-9A6A-7B460619B92D]@0xffffff7f830ba000->0xffffff7f830d2fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[CBAE26D8-0ACB-3C1F-8347-FDCA67EC40B3]@0xffffff7f81239000
         org.hwsensors.driver.CPUSensors(1372.0)[6463CC48-8DEE-3060-8246-CFDE56B4ED3E]@0xffffff7f830d3000->0xffffff7f830d9fff
            dependency: org.netkas.driver.FakeSMC(1372)[D68576F1-B949-3130-9A6A-7B460619B92D]@0xffffff7f830ba000

BSD process name corresponding to current thread: kernel_task
Boot args: npci=0x3000 rootless=0 slide=0 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 15.0.0: Sun Jun 28 00:25:56 PDT 2015; root:xnu-3247.1.36~7/RELEASE_X86_64
Kernel UUID: E6DB13C9-258E-310A-A791-6F9E324C3726
__HIB  text base: 0xffffff8000100000
System model name: iMac13,2 (Mac-FC02E91DDD3FA6A4)

System uptime in nanoseconds: 183012688211
last loaded kext at 183012601290: com.apple.driver.AppleIntelCPUPowerManagementClient	218.0.0 (addr 0xffffff7f81e6b000, size 20480)
loaded kexts:
org.hwsensors.driver.LPCSensors	1372
org.hwsensors.driver.GPUSensors	1372
com.insanelymac.IntelMausiEthernet	2.0.0
org.voodoo.driver.VoodooTSCSync	1.1
org.hwsensors.driver.CPUSensors	1372
org.netkas.driver.FakeSMC	1372
com.apple.driver.AppleIntelCPUPowerManagementClient	218.0.0
com.apple.iokit.IOAHCIBlockStorage	2.8.0
com.apple.driver.usb.AppleUSBEHCIPCI	1.0.1
com.apple.driver.AppleAHCIPort	3.1.5
com.apple.driver.AppleACPIButtons	4.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	163
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleIntelCPUPowerManagement	218.0.0
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.driver.usb.AppleUSBEHCI	1.0.1
com.apple.iokit.IOAHCIFamily	2.8.0
com.apple.iokit.IONetworkingFamily	3.2
com.apple.driver.AppleEFINVRAM	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.driver.AppleEFIRuntime	2.0
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	414
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.iokit.IOUSBHostFamily	1.0.1
com.apple.driver.AppleUSBHostMergeProperties	1.0
com.apple.driver.AppleACPIPlatform	4.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: iMac13,2, BootROM IM131.010A.B05, 4 processors, Quad-Core Intel Xeon E5, 3.7 GHz, 8 GB, SMC 2.11f14
Graphics: AMD Radeon HD 7xxx, AMD Radeon HD 7xxx, PCIe, 3072 MB
Memory Module: BANK0/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Memory Module: BANK1/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Memory Module: BANK2/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Memory Module: BANK3/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: SanDisk SDSSDX120GG25, 120.03 GB
Serial ATA Device: SanDisk SDSSDHP256G, 256.06 GB
Serial ATA Device: ST3250824AS  Q, 250.06 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Turtle Beach Z22 Chat
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB2.0 Hub
USB Device: Razer DeathAdder
USB Device: Gaming Keyboard G110
USB Device: G110 G-keys
Thunderbolt Bus: 
```

winsock, 2015-07-18:
This time my computer KPed 10+ times before booting... Weird.
```
Anonymous UUID:       360F0ED9-3D72-6ECC-2E66-C2F2C451A9E8

Sat Jul 18 14:34:52 2015

*** Panic Report ***
panic(cpu 5 caller 0xffffff80003d7e1a): Kernel trap at 0xffffff800083e3fa, type 13=general protection, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x00000000157fe000, CR4: 0x00000000001626e0
RAX: 0x0000000000000005, RBX: 0xffffff8024055cc0, RCX: 0xffffff802375f400, RDX: 0x0000000000000001
RSP: 0xffffff81139c3c60, RBP: 0xffffff81139c3c90, RSI: 0xffffff8000b24078, RDI: 0xdeadbeefdeadbeef
R8:  0xffffff8022fcd000, R9:  0xffffff8022f11dd8, R10: 0x0000000000000000, R11: 0x0000000000000003
R12: 0xffffff802419f001, R13: 0x0000000000000005, R14: 0xffffff80241784c0, R15: 0x0000000000000008
RFL: 0x0000000000010282, RIP: 0xffffff800083e3fa, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000000, Fault CPU: 0x5, PL: 0

Backtrace (CPU 5), Frame : Return Address
0xffffff811390ddf0 : 0xffffff80002e74d7 
0xffffff811390de70 : 0xffffff80003d7e1a 
0xffffff811390e050 : 0xffffff80003f5553 
0xffffff811390e070 : 0xffffff800083e3fa 
0xffffff81139c3c90 : 0xffffff800083e451 
0xffffff81139c3ca0 : 0xffffff800083ef78 
0xffffff81139c3cf0 : 0xffffff7f830a8f37 
0xffffff81139c3d30 : 0xffffff7f830a8e18 
0xffffff81139c3d70 : 0xffffff7f830a9366 
0xffffff81139c3df0 : 0xffffff7f830ad419 
0xffffff81139c3e80 : 0xffffff8000895add 
0xffffff81139c3ec0 : 0xffffff80008955e6 
0xffffff81139c3f30 : 0xffffff800089029a 
0xffffff81139c3f70 : 0xffffff800089627c 
0xffffff81139c3fb0 : 0xffffff80003d2857 
      Kernel Extensions in backtrace:
         org.netkas.driver.FakeSMC(1372.0)[D68576F1-B949-3130-9A6A-7B460619B92D]@0xffffff7f830a7000->0xffffff7f830bffff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[CBAE26D8-0ACB-3C1F-8347-FDCA67EC40B3]@0xffffff7f81239000

BSD process name corresponding to current thread: kernel_task
Boot args: npci=0x3000 rootless=0 slide=0 

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 15.0.0: Sun Jun 28 00:25:56 PDT 2015; root:xnu-3247.1.36~7/RELEASE_X86_64
Kernel UUID: E6DB13C9-258E-310A-A791-6F9E324C3726
__HIB  text base: 0xffffff8000100000
System model name: iMac13,2 (Mac-FC02E91DDD3FA6A4)

System uptime in nanoseconds: 183729702417
last loaded kext at 183476757152: com.apple.iokit.IOAHCIBlockStorage	2.8.0 (addr 0xffffff7f81809000, size 155648)
loaded kexts:
org.hwsensors.driver.LPCSensors	1372
org.hwsensors.driver.GPUSensors	1372
com.insanelymac.IntelMausiEthernet	2.0.0
org.voodoo.driver.VoodooTSCSync	1.1
org.hwsensors.driver.CPUSensors	1372
org.netkas.driver.FakeSMC	1372
com.apple.driver.AppleIntelCPUPowerManagementClient	218.0.0
com.apple.iokit.IOAHCIBlockStorage	2.8.0
com.apple.driver.AppleAHCIPort	3.1.5
com.apple.driver.usb.AppleUSBEHCIPCI	1.0.1
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleACPIButtons	4.0
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	163
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleIntelCPUPowerManagement	218.0.0
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.iokit.IOAHCIFamily	2.8.0
com.apple.driver.usb.AppleUSBEHCI	1.0.1
com.apple.iokit.IONetworkingFamily	3.2
com.apple.driver.AppleEFINVRAM	2.0
com.apple.driver.AppleEFIRuntime	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	414
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.iokit.IOUSBHostFamily	1.0.1
com.apple.driver.AppleUSBHostMergeProperties	1.0
com.apple.driver.AppleACPIPlatform	4.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.Libm	1
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
Model: iMac13,2, BootROM IM131.010A.B05, 4 processors, Quad-Core Intel Xeon E5, 3.7 GHz, 8 GB, SMC 2.11f14
Graphics: AMD Radeon HD 7xxx, AMD Radeon HD 7xxx, PCIe, 3072 MB
Memory Module: BANK0/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Memory Module: BANK1/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Memory Module: BANK2/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Memory Module: BANK3/DIMM0, 2 GB, DDR3, 1333 MHz, Samsung, M378B5673FH0-
Network Service: Ethernet, Ethernet, en0
Serial ATA Device: SanDisk SDSSDX120GG25, 120.03 GB
Serial ATA Device: SanDisk SDSSDHP256G, 256.06 GB
Serial ATA Device: ST3250824AS  Q, 250.06 GB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB2.0 Hub
USB Device: Razer DeathAdder
USB Device: Gaming Keyboard G110
USB Device: G110 G-keys
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: Turtle Beach Z22 Chat
Thunderbolt Bus: 
```

falanx, 2015-07-22:
I have this issue too. I get like 4-5 kps in a row then it just boots fine. Interestingly I see in your kp the last thing is Thunderbolt Bus or corecrypto. The same exact thing is happing at me too. Like exactly.

falanx, 2015-07-22:
Apple is changing a lot of things under the hood and maybe because our SMBios tell the OS that we should have a Thunderbolt port and we don't it's crashing. My opinion based on assumptions not on any concrete information.

falanx, 2015-07-24:
I was trying a lot of things and figured out I was able to boot without kernel panic when booting without caches. So I moved all my kexts to L/E and rebuilt cache and repaired permissions for S/L/E and for L/E and so far no KP. I tried shutting my comp down and booting it up 4 times and no KP.

winsock, 2015-07-24:
Same, it looks like Clovers kext-injection is completely broken in DP4 anyways. So /L/E and rebuilding the precompiled kernel is the way to go now.

closed 2015-07-24 by winsock