Cozmonat/HWSensors #296: update breaks hwmonitor on mac 10.9.5

harley7991, 2015-08-15:
crash report from console:

Process:         HWMonitor [851]
Path:            /Applications/HWMonitor.app/Contents/MacOS/HWMonitor
Identifier:      org.hwsensors.HWMonitor
Version:         6.18.1394 (1394)
Code Type:       X86-64 (Native)
Parent Process:  launchd [169]
Responsible:     HWMonitor [851]
User ID:         501

Date/Time:       2015-08-15 11:55:21.022 -0400
OS Version:      Mac OS X 10.9.5 (13F1096)
Report Version:  11

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000000

VM Regions Near 0:
--> 
    __TEXT                 000000010c1db000-000000010c246000 [  428K] r-x/rwx SM=COW  /Applications/HWMonitor.app/Contents/MacOS/HWMonitor

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   org.hwsensors.HWMonitor       	0x000000010c1e9eb9 -[StatusItemView drawRect:] + 109
1   com.apple.AppKit              	0x00007fff9111d04f -[NSView _drawRect:clip:] + 3748
2   com.apple.AppKit              	0x00007fff9111b8c4 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 1799
3   com.apple.AppKit              	0x00007fff9111bca0 -[NSView _recursiveDisplayAllDirtyWithLockFocus:visRect:] + 2787
4   com.apple.AppKit              	0x00007fff91119706 -[NSView _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 841
5   com.apple.AppKit              	0x00007fff91248c13 -[NSNextStepFrame _recursiveDisplayRectIfNeededIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:topView:] + 303
6   com.apple.AppKit              	0x00007fff91115e9f -[NSView _displayRectIgnoringOpacity:isVisibleRect:rectIsVisibleRectForView:] + 2828
7   com.apple.AppKit              	0x00007fff910f52da -[NSView displayIfNeeded] + 1680
8   com.apple.AppKit              	0x00007fff9123f3eb -[NSNextStepFrame displayIfNeeded] + 156
9   com.apple.AppKit              	0x00007fff912bd18d -[NSStatusBar _setLengthOfStatusItem:to:] + 428
10  com.apple.AppKit              	0x00007fff910e81e8 NSPerformWithScreenUpdatesDisabled + 65
11  com.apple.AppKit              	0x00007fff912bcf25 -[NSStatusItem _adjustLength] + 498
12  com.apple.AppKit              	0x00007fff91636b2e -[NSStatusItem setView:] + 320
13  org.hwsensors.HWMonitor       	0x000000010c1e996a -[StatusItemView initWithFrame:statusItem:] + 279
14  org.hwsensors.HWMonitor       	0x000000010c200c5c -[PopoverController initWithNibName:bundle:] + 338
15  com.apple.AppKit              	0x00007fff90fad6f4 -[NSCustomObject nibInstantiate] + 368
16  com.apple.AppKit              	0x00007fff90fad520 -[NSIBObjectData instantiateObject:] + 309
17  com.apple.AppKit              	0x00007fff90fac815 -[NSIBObjectData nibInstantiateWithOwner:topLevelObjects:] + 413
18  com.apple.AppKit              	0x00007fff90fa15ce loadNib + 327
19  com.apple.AppKit              	0x00007fff90fa0b6c +[NSBundle(NSNibLoading) _loadNibFile:nameTable:withZone:ownerBundle:] + 288
20  com.apple.AppKit              	0x00007fff90fa095c -[NSBundle(NSNibLoading) loadNibNamed:owner:topLevelObjects:] + 197
21  com.apple.AppKit              	0x00007fff90fa074b +[NSBundle(NSNibLoading) loadNibNamed:owner:] + 357
22  com.apple.AppKit              	0x00007fff90f9c597 NSApplicationMain + 448
23  org.hwsensors.HWMonitor       	0x000000010c1dd164 start + 52

Thread 1:
0   libsystem_kernel.dylib        	0x00007fff8ac5ee6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8815ff08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff88162fb9 start_wqthread + 13

Thread 2:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8ac5f662 kevent64 + 10
1   libdispatch.dylib             	0x00007fff88168421 _dispatch_mgr_invoke + 239
2   libdispatch.dylib             	0x00007fff88168136 _dispatch_mgr_thread + 52

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8ac5ee6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8815ff08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff88162fb9 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8ac5ee6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8815ff08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff88162fb9 start_wqthread + 13

[... 190 lines cut ...]
    thread_set_state: 0

VM Region Summary:
ReadOnly portion of Libraries: Total=174.5M resident=98.4M(56%) swapped_out_or_unallocated=76.1M(44%)
Writable regions: Total=36.6M written=1444K(4%) resident=1652K(4%) swapped_out=0K(0%) unallocated=34.9M(96%)
 
REGION TYPE                      VIRTUAL
===========                      =======
CG backing stores                     4K
CG shared images                    180K
Kernel Alloc Once                     8K
MALLOC                             26.4M
MALLOC (admin)                       32K
STACK GUARD                        56.0M
Stack                              9824K
VM_ALLOCATE                         356K
__DATA                             21.5M
__IMAGE                             528K
__LINKEDIT                         66.7M
__TEXT                            107.8M
__UNICODE                           544K
mapped file                        23.8M
shared memory                         4K
===========                      =======
TOTAL                             313.5M

glaszig, 2015-09-10:
same stacktrace on 10.9.2. not funny.

TiberioG, 2015-09-15:
Same problem too.