Cozmonat/HWSensors #297: 19 RPM when I install Nvidia Web Drivers

jthnrvlls, 2015-08-24:
When I install Nvidia web drivers the GPU fan is stuck at 19RPM and the other hand the PCH Die at 127C. I know that it is a bug but I would like to see all normal. My card is GTX 780 (GK110B) and Ga-Z97X-Sli. 
Thanks.
Version 6.18.1394

cpanderson, 2015-12-10:
Same for me. All other GPU measurements fluctuate so I'm guess they are working.

Latest Nvidia drivers for 10.11.2
GTX 750 Ti
GA-H97M-DH3
Core i5 4960K

winghelo, 2015-12-11:
I have same problem 😭help help
Nvdia driver can show which driver using,
When using OS driver ,rpm show right,
When using Nvdia web driver ,19rpm........
Help help help help......😭😭😭😭😭😭

OS 10.10.5
Nvdia web driver
Inno3D iChill GTX780

jthnrvlls, 2015-12-11:
Definitely this is a mistake. My fan is running in the same speed instead 19 rpm. Gtx's temperature is ok.

kenmadsen, 2016-03-27:
Same here with mac os 10.11.4, nvidia web driver v386.03.06f01 and hwmonitor v6.19.1406.

panbeep, 2016-08-17:
Still no solution? GTX 980Ti showing 19rpm.