Cozmonat/HWSensors #301: wrong temperature

magicnas, 2015-09-17:
on my hack the cpu temp is different than win7. I use HWSensors 6.18.1394, my system is asus p5q deluxe,  xeon e5450. What is wrong? 

[image]
[image]

perfaram, 2015-09-17:
Perhaps OS X is running CPU-demanding background tasks...

magicnas, 2015-09-18:
Cpu load is 1-2%, noting started, on ElCapitan fresh install is the same.

perfaram, 2015-09-21:
Weird... And in the same time, values are still normal. I had problems with completely out of bounds (i.e. -127), but never this kind of apparently correct values.
@kozlek, I loop you in, perhaps you'll know better !

magicnas, 2015-09-27:
can i set TJmax temp manualy?

magicnas, 2015-10-06:
CPUSensors.kext - Contents - Info.plist i set TJmax and now my temp is correct. Thank's