Cozmonat/HWSensors #309: HWMonitor doesn't run on Mavericks

ReddyKilowatt, 2015-11-21:
I'm running Mavericks on a 2012 MBP.
After installing 6.18.1394, double clicking on the HWMonitor.app does nothing.
No screen is displayed, no errors are shown. It's as if I never ran the program at all.

perfaram, 2015-11-21:
Could you please check your log files ? And report here what may be related to HWMonitor. Thanks !

ReddyKilowatt, 2015-11-21:
Where are the log files?

perfaram, 2015-11-24:
Applications/Utilities/Console.app
Open it, start HWMonitor, and see if anything related to HWMonitor appeared.

idaanx, 2016-03-13:
Had the same issue with 6.18 and the new 6.19 release on Marvericks on a hackintosh. 6.17 works without a problem, here's the crash log for 6.19 [HWMonitor_2016-03-13-215012_Z77.crash.zip](https://github.com/kozlek/HWSensors/files/171219/HWMonitor_2016-03-13-215012_Z77.crash.zip)

Marchrius, 2016-04-10:
It doesn't not run becouse of some methods available only in Yosemite or newer.