Cozmonat/HWSensors #310: SuperIODevice: [Fatal] found unsupported chip! ITE sequence ID=0x8628, Winbond sequence ID=0xffff

andreszerocross, 2015-11-23:
Hi Kozlek,
I need help,, 

Can you solve this?? 
I see this log every start with verbose, it looks unsupported chip>

I upload my DSDT, SSDT-x, IOREG, and Screenshot of HW Monitor's sensors

[image]
[image]

andreszerocross, 2015-11-23:
IOREG : https://drive.google.com/file/d/0B8cWaOM9eXf8WS1LRXdXY1l6SmM/view?usp=sharing
DSDT and all SSDT : https://drive.google.com/file/d/0B8cWaOM9eXf8SnRxVFc0ZXU1Tmc/view?usp=sharing

andreszerocross, 2015-11-23:
Owh,, i almost forget about computer specification

Processor : Intel Core i7 6700
Motherboard : Gigabyte Z170-HD3
GPU : Palit Jetstream GTX960 4GB DDR5

kozlek, 2015-11-24:
I see your chip is detected as 0x8628.After short investigation I found it is ITE IT8628E chip. Should be supported by the ITE driver. I'll add it in the next couple of days.

andreszerocross, 2015-11-24:
Thanks before :D

andreszerocross, 2015-11-25:
Hi kozlek,, How can i compile it :D

Can you give me a clue? I am new about this

andreszerocross, 2015-11-26:
Hi Kozlek, it works. But why for RPM value too high???
[image]

Coool, 2015-11-26:
How about Fintek F71809U support! https://github.com/kozlek/HWSensors/issues/238 ;)