Cozmonat/HWSensors #316: Real MacBookAir some sensors at 128oC

rfonseca, 2016-01-18:
I have a problem with my MacBook Air 6,2: intermittently, Memory Module 1, Proximity; Airport Proximity;  Mainboard Proximity; Heatpipe Incoming Air; and Heatpipe Outgoing Air all read 128oC. It seems the real sensors are providing erroneous readings, and the computer becomes unusable: kernel_task goes crazy, and fan goes to 6000+RPM.

My question is: do you know how to, perhaps using FakeSMC, pretend to the OS that these sensors are not crazy, that is, provide fake readings to the kernel?

Thanks!

kozlek, 2016-01-18:
From my experiments in the past, modern OS X kernels  does not control the fans. Seems it is done exclusively by SMC on hardware level. So providing fake values to kernel will not help to solve your issue. But you can try to override fan speed via some 3rd party SMC software.

rfonseca, 2016-01-18:
The main problem is not even the fan, but the fact that the kernel starts one or more idle loop threads to try to prevent the CPU from overheating. These take 200-300% of the 400% CPU cycles (4 cores), making the machine unusable. The idea then would be to not let the kernel see these readings...