Cozmonat/HWSensors #318: ACPISensors causing a panic

vit9696, 2016-03-17:
Hi,
There seems to be a bug with ACPISensors. I seem to be frequently getting a panic while booting my pc, and FakeSMC/ACPISensors are the clear culprits. I have a previous version installed (6.18, but there were no relevant changes since then) via Clover kext injection. And the panic  has been here for years. Would you like to hear more details?

````
Anonymous UUID:       A8516201-D7D8-3D89-8135-878F49216C82

Thu Mar 17 22:52:42 2016

*** Panic Report ***
panic(cpu 4 caller 0xffffff80011d255a): Kernel trap at 0xffffff800163c41a, type 13=general protection, registers:
CR0: 0x0000000080010033, CR2: 0xffffff81e9979000, CR3: 0x000000000ef18000, CR4: 0x00000000001627e0
RAX: 0x0000000000000013, RBX: 0xffffff802a146e40, RCX: 0xffffff8029342680, RDX: 0x0000000000000001
RSP: 0xffffff81f3913bf0, RBP: 0xffffff81f3913c20, RSI: 0xffffff80019270c0, RDI: 0xc0ffee437d81b0ec
R8:  0xffffff80290db800, R9:  0x0000000000000010, R10: 0x0000000000000000, R11: 0x0000000000000000
R12: 0xffffff802a146e01, R13: 0x0000000000000013, R14: 0xffffff8029341c80, R15: 0x0000000000000014
RFL: 0x0000000000010282, RIP: 0xffffff800163c41a, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0xffffff81e9979000, Error code: 0x0000000000000000, Fault CPU: 0x4, PL: 0

Backtrace (CPU 4), Frame : Return Address
0xffffff81e9985df0 : 0xffffff80010df792 mach_kernel : _panic + 0xe2
0xffffff81e9985e70 : 0xffffff80011d255a mach_kernel : _kernel_trap + 0x91a
0xffffff81e9986050 : 0xffffff80011f0233 mach_kernel : _return_from_trap + 0xe3
0xffffff81e9986070 : 0xffffff800163c41a mach_kernel : __ZN7OSArray15initWithObjectsEPPK8OSObjectjj + 0x6a
0xffffff81f3913c20 : 0xffffff800163c471 mach_kernel : __ZN7OSArray13initWithArrayEPKS_j + 0x21
0xffffff81f3913c30 : 0xffffff800163cf98 mach_kernel : __ZN7OSArray14copyCollectionEP12OSDictionary + 0xc8
0xffffff81f3913c80 : 0xffffff7f840cff37 org.netkas.driver.FakeSMC : __ZN15FakeSMCKeyStore6getKeyEPKc + 0x23
0xffffff81f3913cc0 : 0xffffff7f840d1d22 org.netkas.driver.FakeSMC : __ZN13FakeSMCPlugin11isKeyExistsEPKc + 0x28
0xffffff81f3913ce0 : 0xffffff7f840d202b org.netkas.driver.FakeSMC : __ZN13FakeSMCPlugin26addSensorUsingAbbreviationEPKc21FakeSMCSensorCategoryjjfff + 0x10b
0xffffff81f3913d70 : 0xffffff7f840d21f2 org.netkas.driver.FakeSMC : __ZN13FakeSMCPlugin17addSensorFromNodeEP8OSObject21FakeSMCSensorCategoryjj + 0x10a
0xffffff81f3913dc0 : 0xffffff7f840f03b3 org.hwsensors.driver.ACPISensors : __ZN11ACPISensors24addSensorsFromDictionaryEP12OSDictionary21FakeSMCSensorCategory + 0x191
0xffffff81f3913e40 : 0xffffff7f840f0afc org.hwsensors.driver.ACPISensors : __ZN11ACPISensors5startEP9IOService + 0x350
0xffffff81f3913e80 : 0xffffff8001693d2d mach_kernel : __ZN9IOService14startCandidateEPS_ + 0x6d
0xffffff81f3913ec0 : 0xffffff8001693836 mach_kernel : __ZN9IOService15probeCandidatesEP12OSOrderedSet + 0x936
0xffffff81f3913f30 : 0xffffff800168e4ea mach_kernel : __ZN9IOService14doServiceMatchEj + 0x22a
0xffffff81f3913f70 : 0xffffff80016944cc mach_kernel : __ZN15_IOConfigThread4mainEPvi + 0x13c
0xffffff81f3913fb0 : 0xffffff80011ccf77 mach_kernel : _call_continuation + 0x17
      Kernel Extensions in backtrace:
         org.netkas.driver.FakeSMC(1394.0)[2DBA2F88-A115-386A-85D0-F88021A4D2E3]@0xffffff7f840ce000->0xffffff7f840e6fff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5D7574C3-8E90-3873-BAEB-D979FC215A7D]@0xffffff7f825ff000
         org.hwsensors.driver.ACPISensors(1394.0)[BB85F05A-2AF0-3E80-BB37-B8B129AE8AE7]@0xffffff7f840ef000->0xffffff7f840fcfff
            dependency: com.apple.iokit.IOACPIFamily(1.4)[5D7574C3-8E90-3873-BAEB-D979FC215A7D]@0xffffff7f825ff000
            dependency: org.netkas.driver.FakeSMC(1394)[2DBA2F88-A115-386A-85D0-F88021A4D2E3]@0xffffff7f840ce000

BSD process name corresponding to current thread: kernel_task
Boot args: nvda_drv=1 keepsyms=1  

Mac OS version:
Not yet set

Kernel version:
Darwin Kernel Version 15.3.0: Thu Dec 10 18:40:58 PST 2015; root:xnu-3248.30.4~1/RELEASE_X86_64
Kernel UUID: DECC9F02-27D7-31F0-9599-A9EF2873902C
Kernel slide:     0x0000000000e00000
Kernel text base: 0xffffff8001000000
__HIB  text base: 0xffffff8000f00000
System model name: iMac14,2 (Mac-27ADBB7B4CEE8E61)

System uptime in nanoseconds: 2313125562
last loaded kext at 744794748: com.apple.BootCache	38 (addr 0xffffff7f836b4000, size 49152)
loaded kexts:
org.hwsensors.driver.GPUSensors	1394
com.insanelymac.driver.AppleIntelE1000e	3.1.0
org.hwsensors.driver.LPCSensors	1394
org.hwsensors.driver.ACPISensors	1394
org.hwsensors.driver.CPUSensors	1394
org.netkas.driver.FakeSMC	1394
as.vit9696.AppleALC	1.0.3
com.apple.BootCache	38
com.apple.driver.AirPort.Brcm4360	1020.17.1a1
com.apple.driver.usb.AppleUSBEHCIPCI	1.0.1
com.apple.driver.AppleAHCIPort	3.1.8
com.apple.driver.AppleACPIButtons	4.0
com.apple.driver.AppleACPIEC	4.0
com.apple.driver.AppleRTC	2.0
com.apple.driver.AppleHPET	1.8
com.apple.driver.AppleSMBIOS	2.1
com.apple.driver.AppleAPIC	1.7
com.apple.nke.applicationfirewall	163
com.apple.security.quarantine	3
com.apple.security.TMSafetyNet	8
com.apple.driver.AppleUSBMergeNub	900.4.1
com.apple.iokit.IO80211Family	1110.26
com.apple.driver.mDNSOffloadUserClient	1.0.1b8
com.apple.driver.corecapture	1.0.4
com.apple.iokit.IONetworkingFamily	3.2
com.apple.driver.AppleEFINVRAM	2.0
com.apple.driver.usb.AppleUSBEHCI	1.0.1
com.apple.iokit.IOAHCIFamily	2.8.1
com.apple.driver.usb.AppleUSBXHCIPCI	1.0.1
com.apple.driver.usb.AppleUSBXHCI	1.0.1
com.apple.iokit.IOUSBFamily	900.4.1
com.apple.iokit.IOUSBHostFamily	1.0.1
com.apple.driver.AppleUSBHostMergeProperties	1.0.1
com.apple.driver.AppleEFIRuntime	2.0
com.apple.iokit.IOHIDFamily	2.0.0
com.apple.iokit.IOSMBusFamily	1.1
com.apple.security.sandbox	300.0
com.apple.kext.AppleMatch	1.0.0d1
com.apple.driver.AppleKeyStore	2
com.apple.driver.AppleMobileFileIntegrity	1.0.5
com.apple.driver.AppleCredentialManager	1.0
com.apple.driver.DiskImages	417.2
com.apple.iokit.IOStorageFamily	2.1
com.apple.iokit.IOReportFamily	31
com.apple.driver.AppleFDEKeyStore	28.30
com.apple.driver.AppleACPIPlatform	4.0
com.apple.iokit.IOPCIFamily	2.9
com.apple.iokit.IOACPIFamily	1.4
com.apple.kec.pthread	1
com.apple.kec.corecrypto	1.0
com.apple.kec.Libm	1
Model: iMac14,2, BootROM IM142.0118.B12, 4 processors, Intel Core i7, 3.50 GHz, 16 GB, SMC 2.15f2
Graphics: NVIDIA GeForce GTX 650, NVIDIA GeForce GTX 650, PCIe, 2047 MB
Memory Module: BANK0/DIMM1, 8 GB, DDR3, 2400 MHz, Kingston, KHX2400C11D38GX
Memory Module: BANK1/DIMM1, 8 GB, DDR3, 2400 MHz, Kingston, KHX2400C11D38GX
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x111), Broadcom BCM43xx 1.0 (7.21.94.136.1a1)
Bluetooth: Version 4.4.3f4 16616, 3 services, 27 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
PCI Card: NVIDIA GeForce GTX 650, Display Controller, Slot-1
Serial ATA Device: INTEL SSDSC2BW120A4, 120.03 GB
Serial ATA Device: ST1000DM003-1CH162, 1 TB
Serial ATA Device: ASUS    BW-14D1XT
Serial ATA Device: ST2000DL001-9VT156, 2 TB
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: USB 3.0 Bus
USB Device: ASM107x
USB Device: BRCM20702 Hub
USB Device: Bluetooth USB Host Controller
USB Device: ASM107x
USB Device: USB Keyboard
USB Device: USB2.0 Printer (Hi-speed)
USB Device: FA GoFlex Desk
Thunderbolt Bus: 
````

usr-sse2, 2016-03-25:
I never had such a panic before. Some days ago I read this issue report – and today I had this panic.

poof86, 2016-12-22:
Also had the same panic today after updating. I have fakesmc and sensors in my bios, but now also added them to my Efi partition for injection (I'm using Ozmosis). I was hoping the newer version would overwrite the previous kexts and not cause any problems.