Cozmonat/HWSensors #328: Real MBP 2015 trouble reading fans

johnkeates, 2016-07-08:
There seems to be an issue with the fan speed readings, over time it adds new random entries with what seems like wrong outputs: http://imgur.com/a/rOlsE

natanz, 2016-07-08:
Ok, I check it, I have access to one like this.

kozlek, 2016-07-08:
Ok, I'll check it, I have access to mac like this

johnkeates, 2016-07-08:
It basically grows, and when hwmon is restarted it returns back to 1 entry (but still wrong), and then it grows again. http://imgur.com/a/kTRT5

kozlek, 2016-07-09:
Just uploaded build that probably fixing this issue: http://hwsensors.com/public/HWMonitor.app.zip

Please check how it works now.

johnkeates, 2016-07-09:
Yep, that fixed it here. I'll keep an eye on the strings in the menu for a bit to be sure.

kozlek, 2016-07-10:
This should be ok now. Fixed in: 4df0e7abc56c7eb8869ec20fccb96a80c052bf6a

closed 2016-07-10 by kozlek

johnkeates, 2016-07-10:
Quick question: looking at the sources (I haven't really dug into the HWSensors sources, nor do I use Objective-C a lot at the moment), was the fix somewhat like: 

- make sure no useless/illegal characters are used (filter non a-zA-Z0-9 stuff)
- make sure to not dispatch it again and again (dispatch once to not duplicate it)

Or is it more complex than that?

kozlek, 2016-07-10:
> make sure no useless/illegal characters are used (filter non a-zA-Z0-9 stuff)

It is correct

> make sure to not dispatch it again and again (dispatch once to not duplicate it)

dispatch_once is used to build character set only once. 

The problem was with the fan name. Fan name also used as an unique identifier. On real mac fan name appears not to be ended with the /0 char. "FDS" structure provides buffer with the name that has fixed length. And on Macs this buffer is filled with the space chars up to the length. So I just take the buffer, use it as a string with the fixed length and trim trailing spaces or other illegal chars.