Cozmonat/HWSensors #337: Lost fan control after update to 6.21

aszahran, 2016-08-22:
After update form 6.16 to 6.21, Hovering over fans does not bring the fan control dialogue.

aszahran, 2016-10-06:
I tested all versions, and the latest release with working fan controls on MacBook Pro 15 (macOs Sierra) is 6.17, though this version has a weird yellow background since Sierra update.

aszahran, 2017-01-16:
As of version 6.24, there is still no fan control.

kozlek, 2017-01-16:
I am sorry but I disabled fan control on Macs. It was couple of reports about improper fan behavior required SMC reset to fix the issue. 

This is latest version with fan control enabled: [http://hwsensors.com/public/HWMonitor.app.zip](url)

ilanbiala, 2017-03-27:
The above URL is broken, are current builds available from http://hwsensors.com/releases equipped with fan control?