Cozmonat/HWSensors #342: Any plan to upgrade RadeonSensors?

sarim, 2016-10-14:
Seems like RadeonSensors is based on a older linux driver. New generations are missing. Any plan to upgrade it same level to linux amdgpu driver? torvalds/linux/drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c has support for latest POLARIS chips.

josbeir, 2016-12-20:
+1 RX480 is working perfectly but no gpu sensor data :(

JimBartlett, 2017-11-10:
+1
I have a RX580 running on High Sierra 10.13.1 no GPU temp or fans etc. Was previously working on 10.12.6. HWMonitor Version 6.26-322-g1b20eab7.1765.

Would be happy to trial a updated version or provide I/O info etc.

liv3010m, 2018-01-31:
Same here, RX 580 on 10.13 and missing GPU temps and fan speed.

dmatora, 2018-03-12:
@sarim @josbeir @JimBartlett @liv3010m enjoy

liv3010m, 2018-03-12:
OMG! nice!!!
Hopefully a new compiled version of GPUSensors is released soon with your code update!
Thank you so much!

dmatora, 2018-03-12:
@liv3010m you can use High Sierra build
Try this
[HWSensors.6.26-20-g9f76b606.1463.Binaries.dmg.zip](https://github.com/kozlek/HWSensors/files/1800955/HWSensors.6.26-20-g9f76b606.1463.Binaries.dmg.zip)

sarim, 2018-03-12:
any binary for sierra?

liv3010m, 2018-03-12:
Thank you! I tried it but nothing changed (kext got loaded as kextstat reports). Does it have any other requirement?

dmatora, 2018-03-12:
@liv3010m what is your device id?

liv3010m, 2018-03-12:
@dmatora it's 0x67df (It's a Sapphire Pulse RX 580)

dmatora, 2018-03-12:
@liv3010m 0x67DF is supported, you likely don't have kexts installed properly (maybe have multiple instances of kexts loading from S/L/E, L/E, C/k/O

dmatora, 2018-03-13:
@sarim here you go
[HWSensors.6.26-21-gddf53ec1.1464.Binaries.dmg.zip](https://github.com/kozlek/HWSensors/files/1804942/HWSensors.6.26-21-gddf53ec1.1464.Binaries.dmg.zip)

liv3010m, 2018-03-13:
@dmatora thanks for your reply!
I load my kexts from EFI partition, from the 10.13 and the Other directories.

`Livs-Hackintosh:~ ro$ kextstat -l | grep -y fakesmc
   42    4 0xffffff7f87795000 0x11000    0x11000    org.netkas.driver.FakeSMC (1765) 9EEC6103-EEBA-34B3-B9AF-8FE333C43E90 <11 7 5 4 3 1>
Livs-Hackintosh:~ ro$ kextstat -l | grep -y sensors
   43    0 0xffffff7f877a6000 0x5000     0x5000     org.hwsensors.driver.CPUSensors (1765) AB5DE333-8AC8-3549-86E5-C5647AACA61F <42 7 5 4 3>
   55    0 0xffffff7f877ac000 0x8000     0x8000     org.hwsensors.driver.ACPISensors (1765) FA4DF6C0-0D3A-3CE2-8569-0AB2C84053E7 <42 11 7 5 4 3>
   66    0 0xffffff7f877b6000 0x12000    0x12000    org.hwsensors.driver.GPUSensors (1765) 714EA7DC-0CED-3374-AC08-B1D6EC33D53B <42 12 11 7 5 4 3>
   67    0 0xffffff7f877cb000 0xd000     0xd000     org.hwsensors.driver.LPCSensors (1765) 93417250-FFB4-39DA-8B25-F164DE72E120 <42 12 11 7 5 4 3>`

JimBartlett, 2018-03-14:
You beauty! Bonza! Thanks for your work on this. Working perfectly on High Sierra 10.13.3 (17D102) using HWSensors.6.26-21-gddf53ec1.1464.Binaries...
I'm using a Sapphire RX 580 Nitro+ OC 8GB (DeviceID: 0x67df RevisionID 0x00e7) and it is reporting the GPU temp fine. Temp normally sits at around 46-48C and have had it up to 73C running the Heaven benchmark.

karanwadhwa, 2018-05-28:
@dmatora MSI RX480 8GB on 10.13 and missing GPU temps and fan speed.
Chipset Model:       Radeon Pro 480/575
Slot:	                        PCI Slot 0
VRAM (Total):	        8 GB
Vendor:	                AMD (0x1002)
Device ID:	        0x67df
Revision ID:	        0x00c7
Using the kexts you provided above in HWSensors.6.26-21-gddf53ec1.1464.Binaries.dmg.zip
I know the kexts are loading up fine

> kextstat -l | grep -y sensors
   87    0 0xffffff7f80f53000 0xe000     0xe000     org.hwsensors.driver.LPCSensors (1765) 93417250-FFB4-39DA-8B25-F164DE72E120 <16 12 11 7 5 4 3>
   92    0 0xffffff7f80e84000 0x8000     0x8000     org.hwsensors.driver.ACPISensors (1765) FA4DF6C0-0D3A-3CE2-8569-0AB2C84053E7 <16 11 7 5 4 3>
   94    0 0xffffff7f80d3e000 0x5000     0x5000     org.hwsensors.driver.CPUSensors (1765) AB5DE333-8AC8-3549-86E5-C5647AACA61F <16 7 5 4 3>
  117    0 0xffffff7f80d45000 0x12000    0x12000    org.hwsensors.driver.GPUSensors (1765) 714EA7DC-0CED-3374-AC08-B1D6EC33D53B <16 12 11 7 5 4 3>

towards the end of `bdmesg`:
> 10:100  0:012  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\Other
10:100  0:000  Extra kext: EFI\CLOVER\kexts\Other\LPCSensors.kext (v.1464)
10:168  0:067  Extra kext: EFI\CLOVER\kexts\Other\GPUSensors.kext (v.1464)
10:188  0:019  Extra kext: EFI\CLOVER\kexts\Other\AppleALC.kext (v.1.2.2)
10:236  0:047    |-- PlugIn kext: EFI\CLOVER\kexts\Other\AppleALC.kext\Contents\PlugIns\PinConfigs.kext (v.1.0.0)
10:469  0:233  Extra kext: EFI\CLOVER\kexts\Other\WhateverGreen.kext (v.1.1.4)
10:499  0:030  Extra kext: EFI\CLOVER\kexts\Other\Lilu.kext (v.1.2.2)
10:529  0:029  Extra kext: EFI\CLOVER\kexts\Other\IntelMausiEthernet.kext (v.2.3.0)
10:548  0:019  Extra kext: EFI\CLOVER\kexts\Other\FakeSMC.kext (v.1765)
10:575  0:026  Preparing kexts injection for arch=x86_64 from EFI\CLOVER\kexts\10.13
10:575  0:000  Extra kext: EFI\CLOVER\kexts\10.13\AppleALC.kext (v.1.2.2)
10:634  0:059    |-- PlugIn kext: EFI\CLOVER\kexts\10.13\AppleALC.kext\Contents\PlugIns\PinConfigs.kext (v.1.0.0)
10:857  0:222  Extra kext: EFI\CLOVER\kexts\10.13\WhateverGreen.kext (v.1.1.4)
10:871  0:014  Extra kext: EFI\CLOVER\kexts\10.13\Lilu.kext (v.1.2.2)
10:883  0:012  Extra kext: EFI\CLOVER\kexts\10.13\IntelMausiEthernet.kext (v.2.3.0)
10:896  0:012  Extra kext: EFI\CLOVER\kexts\10.13\FakeSMC.kext (v.1765)
10:923  0:027  SetScreenResolution: 1920x1080 - already set
10:923  0:000  Custom boot is disabled

also from what i could catch in verbose boot, after ACPISensors, CPUSensors, PTID?Sensors have started this is the exact output i get from RadeonSensors

> RadeonSensors  (pci1):  [Fatal] unknown card 0x67df
RadeonSensors  (pci1):  found ATI Radeon ID: 0x67df, ATOM BIOS: 113
RadeonSensors  (pci1):  using device-id to resolve temperature sensor tpe
RadeonSensors  (pci1):  started

I'm fairly new at this and not entirely sure what I'm doing, maybe I'm forgetting something?