Cozmonat/HWSensors #355: How can I get ALL OF CORES Freq for 2 CPU

rio-redstone, 2017-03-26:
Now I use HP xw6600 (Intel 5400 chipset, **Xeon 5450 x2**, 32GB RAM, AMD Radeon RX460) with macOS Sierra. 
first of all, you know, Xeon 5450 has 4 cores, so my hack has 8 cores in total.
On your HWMointor, "Temperatures" shows for all of cpu cores', 8 cores of 2 cpu. but "Frequencies" shows half of cpu cores, means only 4 cores. in attached below, I think you can see that my hack has 8 cores working, and HWMonitor shows only 4 cpu core frequencies, beside 8 cpu core temperatures. 
I would like to know all of cpu core frequencies, 8 core of 2 cpu. how can I get all information?? 
do you have any suggestion??

thanks

[image]
[image]

rio-redstone, 2017-06-04:
I think no one is interested in this topic, but I fix this issue.
in CPUSensors.cpp, when a task collects frequencies, for-loop is like,

`for (uint32_t i = 0; i < cpuid_info()->core_count; i++)`

cpuid_info() injects only 1 cpu. so core_count sets only 4 (means half of all cores).
so my Hack has 2 cpus, 8 cores, I changed the line like,

`for (uint32_t i = 0; i < 8; i++)`

I think that 
`for (uint32_t i = 0; i < coreCount; i++)`
will work well and is better than my method.

I hope this will helps someone.

red-green, 2017-08-10:
What if you read the number of CPU core temperatures that were reported and used that number as a reference?