Cozmonat/HWSensors #357: Feature request: S.M.A.R.T. Error rate calculator for Seagate drives

jammi, 2017-03-29:
Since Seagate (ST-somethings) report the total amount of reads and seeks in a format that includes the total amount and errors in one, they always have insanely high raw numbers. To make sense of them, it's required to parse the binary value of the 48-bit number and split the first 16 bits as errors and last 32 bits as total. So what I suggest is displaying something like **(errors: \[number], total: \[number])** rather than, or in addition to the raw value.

Here's an article about the subject I found: http://sgros.blogspot.com/2013/01/seagate-disk-smart-values.html

AFAIK, the fields this'd apply to would be "Raw Read Error Rate" and "Seek Error Rate".

kozlek, 2017-03-29:
There is no problem to add 16/24 format support. But this should be configured via configuration file. It is possible to apply this rule to specific hdd that is matched by name. Do you have Seagate hdd?

jammi, 2017-03-29:
Three of the drives are Seagate, two different models.

kozlek, 2017-03-29:
Please, share your HDD names here. I will add them to config. I need the names as they are displayed in HWMonitor if possible.

jammi, 2017-03-29:
`ST2000DL001-9VT156` and `ST4000DM000-1F2168`

jammi, 2017-03-29:
And here are screenshots of the drives in question: 
[image]
[image]
[image]

kozlek, 2017-03-29:
You can try [this]( http://hwsensors.com/public/HWSensors.6.24-4-g0d00fecf.1425.Binaries.dmg)

jammi, 2017-03-29:
Thank you, it works.

closed 2017-03-29 by jammi