Cozmonat/HWSensors #358: Pascal GPU Sensors

sincohen, 2017-04-12:
Hello,
Pascal drivers for macOS just released. However GPUSensors.kext causes a KP. Can we add Pascal support or at least suppress the KP?
Thank you

kozlek, 2017-04-12:
Can you check this one?

 http://hwsensors.com/public/HWSensors.6.25-12-g23039705.1438.Binaries.dmg

duffs, 2017-04-12:
hi ! from kernel log
kernel: (kernel) RadeonSensors (pci1): [Fatal] unknown card 0x67ef
kernel: (kernel) RadeonSensors (pci1): found ATI Radeon ID: 0x67ef, ATOM BIOS: C99401
kernel: (kernel) RadeonSensors (pci1): using device-id to resolve temperature sensor type
kernel: (kernel) RadeonSensors (pci1): started

started and no kp but no sensors..

Dids, 2017-04-13:
@kozlek I can confirm those work nicely. No more KPs, but also getting good GPU sensor readings.

JRDNb, 2017-04-16:
@kozlek I have installed the files from the link above to S/L/E does it belong anywhere else?
I am unable to get temp readings from my 1050ti. Was previously working with 750ti

EricTheRed1, 2017-04-17:
Almost everything is working the new new kext files.  My GPU core freqency stays at 54 Mhz on my 1080.

lignicolos, 2017-04-17:
I'm seeing the same 54 MHz for gpu core on my MSI 1060.

Dids, 2017-04-17:
Yeah, GPU Core shows 54 Mhz on a 1070 here.
GPU Memory does jump to 8.01 Ghz when I launch a game though, then back to 810 Mhz when idle.

Dids, 2017-04-17:
@JRDNb Be sure to remove any other FakeSMC/sensor kexts before installing the new ones, and as always, repair permissions and rebuild the kext cache.

JRDNb, 2017-04-17:
@Dids Thanks! I ended up having to remove kexts that had FakeSMC_ infront of the kext name. That seemed to do the trick. However, I now have my GPU PWM showing 100% and a new fan CPUFan#1 which shows 0rpm. And, the other reported 54Mhz for frequency. Temps are showing up though which is nice!

demonknight9, 2017-04-17:
Is there a new fakeSMC with the plugins already ready to install? I can't seem to get the readings to work on my 1070? Maybe I am doing something wrong though so if anyone can be so kind as to post the new FakeSMC already compiled with the new binaries I really appreciate it :)

demonknight9, 2017-04-17:
Do I just create a plugins folder inside of the FakeSMC kext and then install the kext to extensions? I am really lost with this. Need my GPU frequencies and temps back. Someone help please :(

JRDNb, 2017-04-18:
@demonknight9 I used kext utility (kextbeast also works) and installed to the S/L/E and L/E folders I did both for good measure. I believe they only need to go in L/E

demonknight9, 2017-04-18:
Thanks for your help JRDNb. It doesn't look like it's working for me. I have one of the GTX 1070's that has micron memory and I updated the card's bios. It was working before that but not anymore.
I will just have to wait for fakesmc and see if that fixes the issue.

xlla, 2017-04-24:
I am using gtx 1060 oc, it work with HWSensors.6.25-12-g23039705.1438.Binaries.dmg, but gpu frequency was 54Mhz.
I have dual-gpu, one is hd630 , another is gtx 1060, both enabled, does this sensor support multi-gpu ?

demonknight9, 2017-04-26:
I have the same issue with single GPU. It's a bug I think, I did manage to get the temps and memory running properly. Frequency is stuck on 54MHz.

Lukypie, 2017-05-13:
I've wrongly installed the older drivers, so all I get is a kernel panic. I used the .app installer, but I am unable to boot up successfully again **even if I remove the old kexts from S/L/E**, what am I doing wrong? The only FakeSMC kext is the one I used for the installation without nothing in EFI/CLOVER

Hunk89, 2017-05-25:
Hi,

is the Frequency READING 54MHz or is it 54MHz in Reality?
I get a kernel panic after typing my password.

How is the Performance? 

THX
Hunk

EricTheRed1, 2017-05-25:
No, the frequency is not 54Mhz.  It's very wrong.  It always displays that.  Make sure you are using the GPU sensors kext that is linked above to avoid a kernel panic.  Pascal performance is a lot better.  My 1080 uses a lot less power than my 980Ti also.  Less power = less heat.

edgard, 2017-05-27:
Same here, 1080TI FE shows 54Mhz. Everything else works.

timcharper, 2017-05-31:
I've put all of the kexts from the above linked dmg in both /System/Library/Extensions (via kext tool) and /Volumes/EFI/EFI/CLOVER/kexts/Other (confirmed to be loading other kexts here, so I'm confident this loading mechanism works). However, I still get the following kernel panic with an Nvidia 1060 card:

[image]

koush, 2017-06-22:
@kozlek can confirm your fix on my system as well. 1080ti was causing kernel panics, installed the build you posted in https://github.com/kozlek/HWSensors/issues/358#issuecomment-293537614 and it resolved the issue. Temperature reporting works as well.

I noticed that this build isn't on http://www.hwsensors.com/releases

I came here to report the issue, but found it was already fixed.

dekefake, 2017-06-23:
I own a GTX 1050ti, and im using the build 6.25-12-g23039705.1438.
The system boots fine, but i cant get any readings. kexts are installed in EFI partition, with Clover.

WeWin55, 2017-06-25:
I own a GTX 1050 Ti OC (Gigabyte). Ive installed the GPUSensors in /S/L/E with Version 6.25.1426 and the system boots fine, but i cant see any information about the card.
@dekefake Do you have any tips?

dekefake, 2017-06-26:
WeWin55 exactly, you need to 1438 build for Pascal cards :)

dskvr, 2017-09-23:
Looks like the domain registry bill wasn't paid `HWSensors.6.25-12-g23039705.1438.Binaries.dmg` is gone with the wind. Would it make sense to upload those binaries as releases on github? 🤔

AnTran131, 2017-10-06:
i can't boot card graphic 1060 on MSI ge72vr 7rd