Cozmonat/HWSensors #360: Seagate newer model drives the maximum temperature is now at 60 degrees Celsius

TheDreamer, 2017-04-12:
As noted in FAQ: http://knowledge.seagate.com/articles/en_US/FAQ/193771en

Their newer model drives permit a maximum temperature of 60 degrees C, which makes the alerts for my external, passive air-cooled (though I have a USB fan sitting on top of it) Seagate drive (Backup Plus Hub), which normally sits around 50 degrees C, kind of annoying.

HWMonitor shows its USB Product Name of "Backup+ Hub BK", while smartctl's info gives:

Model Family:  Seagate Archive HDD
Device Model:  ST8000AS0002-1NA17Z

The datasheet: http://www.seagate.com/files/www-content/product-content/hdd-fam/seagate-archive-hdd/en-us/docs/archive-hdd-dS1834-3-1411us.pdf

Shows that its max operating temperature is 60 degrees C.

Can the alert thresholds be changed?

kozlek, 2017-04-12:
Could you send me IORegistryExplorer dump?

TheDreamer, 2017-05-09:
Is this right?

[numac-ioreg.zip](https://github.com/kozlek/HWSensors/files/987658/numac-ioreg.zip)

kozlek, 2017-05-19:
Yes. But in txt it's impossible to analyze it. Just run IORegistryExplorer and go to File->Save and upload here .ioreg file