Cozmonat/HWSensors #362: Feature request - eGPU monitoring ability

fireheadman, 2017-04-30:
This may be an impossible request, but wanted to field it...  never hurts to ask.

There is a small community of users that attach eGPUs (external GPU) to their macs (whether a mac mini/pro/macbook...etc).  The majority use the latest Nvidia cards.  The eGPU's are connected via thunderbolt2/3 cables.

Would like to know if it is possible to query temperatures across thunderbolt connections?  If so, could you  make this available in your HWSensors pkg?

fireheadman, 2017-05-01:
Stats such as GPU core Temp, Fan Speed, possibly Voltage to start would be awesome!

tanshihaj, 2017-06-03:
Yes, it will be really great feature.

DANgerous25, 2017-06-13:
Hi there, I would also love to see this feature. For what it's worth, HWSensors is a really nice app btw. Thank you.

ikir83, 2017-11-29:
+1 it would be very handy for eGPU

sergiorru, 2017-12-27:
I could’nt agree more, this feature will be awesome!

ddqp, 2017-12-28:
is there any update on this topic?

hunglc007, 2018-01-09:
yes, i will be great feature