Cozmonat/HWSensors #363: FakeSMC_LPCSenors and "W_PUMP" Header not Working

nnapior, 2017-05-06:
When I have the FakeSMC_LPCSensors.kext installed my motherboard's (Specs Below) water pump header is completely ignored in HWMonitor and not only that but the port completely shut off. This lead to temperature problems because on boot into Mac OS my water pump would shut off completely. I determined it to be the FakeSMC_LPCSensors.kext because by only removing that kext the port works fine. My motherboard has 5 fan headers and the single "W_PUMP" (water pump) port. In HWMonitor only the 5 ports were available. Hope this is helpful.

TL;DR: my motherboard has 6 fan headers (5 regular fan headers and 1 W_PUMP header). FakeSMC_LPCSensors.kext causes the W_PUMP header to shut off completely and not show up in HWMonitor.

Hackintosh Specs:
Motherboard: Asus z170-Deluxe
CPU: Intel i7-6700k
RAM: Corsair LPX 16Gb (2400Mhz XMP)
GPU: Nvidia GTX 1080
Boot SSD: Samsung 960 Pro 512Gb (partitioned half for windows half for Mac OS) NVMe SSD
Operating System: OSX Sierra 10.12.4
Custom Loop (all fans are running in DC mode and water pump is running in PWM)

Hope to see a fix, awesome work you guys put out. Thanks.

osxfr33k, 2017-06-21:
Could catch I have the Asus Maximux VIII Hero Alpha had to resort to installing two builds below the latest one saw the same issue as you you excessive temperatures exceeding to 90c+ even at idle.  I did n;t realize the water pump header was shutoff.  Hopefully this can be fixed soon since this is the second reporting.   I would think anyone using that header would see the same issue.   Did you just switch that to a different header and ignore that one?  Might be hard to fine one close to CPU short wire from cooler to header.  Can we use 12 volts from Power suppler with a connector converter or we need to place this on a header?

nnapior, 2017-07-11:
Sorry for the late reply. As you can see in my post, I solved it by removing FakeSMC_LPCSensors.kext which does remove some of the functionality of HWMonitor, but enables the use of the water pump port. Hope this helps.

osxfr33k, 2017-07-22:
Yes but I prefer to see temps so I reverted to 6.21 until Kozlek fixes this.