Cozmonat/HWSensors #365: Is this code portable to linux?

shkronjax, 2017-06-04:
this code runs great on a mac...with the exception of the radeon sensor...How would I go about compiling this on linux?

red-green, 2017-06-22:
https://github.com/groeck/lm-sensors/

shkronjax, 2017-06-23:
I already have lm-sensors and xsensors interface installed. However, lm-sensors lacks support for mac 5.1 (mid 2010) when it comes to sensor names and some other features such as voltage etc...Since hwsensors opens up in xcode, I was wondering if anyone out there has the knowledge of cross-compiling or even making hesensors compatible with Ubuntu or other Linux flavors. The question is mainly directed towards the developers of this project. Thanks red-green for your input.

red-green, 2017-06-24:
I think the problem is that this kernel extension is designed to integrate with fakeSMC which there didn't seem to be any reason to make for Linux. You'd have to see if anyone has done so, but I think the kernel extension differences between Mac and Linux might make it impossible.

shkronjax, 2017-06-24:
Unhappily, I agree. Thank You for your answer.

closed 2017-06-24 by shkronjax