Cozmonat/HWSensors #372: HWSensors causing kp in X299 i9-7900

surfinchina, 2017-08-11:
Hi,
I get a kp if I try to run the HWSensors kexts, or as plugins in fakeSMC.
I've tried the latest versions of the kexts and fakeSMC but get the same kp.
Is there a timeline as to when this CPU will be supported and how can I help :)
surfi
[image]

anonym323, 2017-11-15:
Would also be very thankful for an update for x299 builds

toohotz, 2018-01-15:
@surfinchina this PR https://github.com/kozlek/HWSensors/pull/387 was made in effort to bring support but I cannot boot myself with the kexts, stuck on a black screen after a timeout on **AppleCPICPU**