Cozmonat/HWSensors #382: Repeated requests for helper tool installation

pointlessone, 2017-10-19:
A password prompt is displayed multiple times during launch of HWMonitor, changes to fan control configuration and just once in a while. The prompt says "HWMonitor is trying to install a new helper tool"

Console has this entry afterwards: Helper service of "org.hwsensors.HWMonitor" failed initial static codesignature check.

HWMonitor: 6.26.1440
macOS High Sierra

Baekalfen, 2017-12-16:
Did you solve the problem?

pointlessone, 2017-12-17:
No, not really. I tried different forks. Some of them were a bit better (didn't ask for password as much or as often) but I ultimately gave up.

darthmaulbr, 2018-02-03:
I have the same issue.

Baekalfen, 2018-02-03:
My suspicion is, that the privileged helper tool doesn't get set up properly. Then it will request the password to create the helper tool every time it calls a privileged function.

I haven't had the time to look at it, but it could be because the certificates don't match between the two programs