Cozmonat/HWSensors #388: Unable to install FakeSMC.kext onto real MacBook Pro 11,3

computoman, 2018-01-03:
Hello,

I have been attempting to install FakeSMC onto my MacBook in hopes of fixing a crashing issue that I am having. When I use windows on my MacBook, the system runs fine, but under OSX, the system crashes without a panic report. I am guess that it works perfectly fine under windows because the system isn't utilising the SMC in the same way. I have already tried disabling the graphics switching, and this does not fix the problem.

When I install the FakeSMC.kext and set correct permissions, the kext does not load, and under the system report is says 'loadable:no'. I have turned off SIP. There isn't much information regarding installing FakeSMC on real Macs. Do I need to first disable the native SMC driver? Any suggestions would be greatly appreciated.

Ben

toohotz, 2018-01-18:
@computoman to make a long post relatively short, `FakeSMC` was never intended to be used on real Macs as they do not / should not require it.  I'd suggest you look into what it is and how it is used before you attempt to use it on your MacBook.

tofifee, 2018-03-20:
Do you really want to brick your MacBook?