Cozmonat/HWSensors #396: Support for Nvidia GT 1030

DevAndArtist, 2018-02-15:
Hi there, I think it's fair to say that some people from the CustoMac community including myself would really appreciate if someone could add support for the low budget pascal cards like the GT 1030.

My GT 1030 is from MSI and is passive cooled. Sometimes the card is getting really hot but there is no way to tell what the temperature is (pure macOS system). I only can "hope" that the card is under 80C° under stress. The latest kexts from `https://github.com/kozlek/HWSensors/tree/6.26` do not support this card, or at least the support is bugged, but I cannot tell for sure.

I already tried everything:
* /EFI/CLOVER/kexts/Other/
* /S/L/E/
* /L/E/

The other sensors load from any of these folders and work just fine, but only GPU sensors are missing.

dmatora, 2018-03-01:
@DevAndArtist enjoy

DevAndArtist, 2018-03-02:
@dmatora thank you, but how do I get it to work? I tried to compile the project in different ways. I placed the compiled `GPUSensors.kext` in clover, it didn't work, then in /S/L/E/ with the same result. Actually I replaced all the kext's that I had with the compiled ones and it broke my system because the `FakeSMC.kext` wasn't working as expected. I was stuck for 5 hours searching for a solution until I restored a previous version of `FakeSMC.kext` and my system finally could boot. Lesson learned `FakeSMC.kext` is mandatory and should not be corrupted. That said, I'm probably doing something wrong and unable to create working kext's.

@kozlek mind a new release for the current state of the project?

chris1111, 2018-03-02:
The release V6 26  1440 works good here High Sierra 10.13.3

DevAndArtist, 2018-03-02:
@chris1111 for me `HWMonitor` nor `iStat Menu` does not recognize the temperature for my MSI GT 1030. `iStat Menu` however is able to display FPS for the graphics card. I hope the fix from @dmatora will finally allow me to view the current temperature.

chris1111, 2018-03-02:
I will test right now to my  Zotac GeForce GT 1030 2 gig and I let you know

chris1111, 2018-03-02:
Post upper  edited

DevAndArtist, 2018-03-02:
@chris1111 Yeah everything works great except for the temperature for that card.

dmatora, 2018-03-02:
>Actually I replaced all the kext's that I had with the compiled ones and it broke my system because the FakeSMC.kext wasn't working as expected
@DevAndArtist looks like you've failed to install kexts properly

@chris1111 does it now show temperature for your GT 1030?

DevAndArtist, 2018-03-02:
@dmatora mind to put a small instruction how to compile the project properly if you start by checking out the repo?!

dmatora, 2018-03-02:
@DevAndArtist just selecting "Build kexts" on tasks and pressing play button builds the kexts under the Build subdirectory

DevAndArtist, 2018-03-02:
@dmatora did that, but it won't be signed nor does the kext has a version tag attached to it. I fail to run create installer though. 

I also tried to select the `GPUSensors` target and enable automatic signing in Xcode 9, but this is the current version I have in /S/L/E which doesn't seem to run.

dmatora, 2018-03-02:
you don't really need to sign it, to run it

dmatora, 2018-03-02:
@DevAndArtist so does temperature sensor works for you now?

DevAndArtist, 2018-03-02:
@dmatora yes I tested the card playing a game for 30 minutes it gets up to 82C°, which is fine for a passive cooled card. Without 3D tasks it's around 35-45C°. We now only need NVidia to fix their drivers. :) Thank you again.

dmatora, 2018-03-02:
@DevAndArtist my pleasure

dmatora, 2018-03-02:
@chris1111 you might want to update your build

DevAndArtist, 2018-03-02:
@dmatora btw. is it normal that I now have an entry for GPU PWM? It does not make sense. It's running at 100% or 100rpm (depends on the app I'm using), but there is no fan for that.

dmatora, 2018-03-02:
@DevAndArtist kext doesn't have dedicated support for at least that particular passive card, it just thinks all 1030 have fans.

DevAndArtist, 2018-03-02:
@dmatora okay, I just wanted to know. I mean theare are 1030 cards with fans out there so it's fine, I'll just filter that entry out in iStat.

chris1111, 2018-03-02:
I am not able to compilling the source with a Xcode 9.2 (9C40b) Error missing version.h

DevAndArtist, 2018-03-02:
@chris1111 you have to use `HWSensors.xcworkspace` not `.xcodeproj`.

chris1111, 2018-03-02:
@Adrian Z. Trying both, same error

DevAndArtist, 2018-03-02:
@chris1111 which build scheme are you using? `Build kext` is the right one.

chris1111, 2018-03-02:
@Adrian Z. Please post pics of the selection I am not see where

DevAndArtist, 2018-03-02:
[image]

Then hit the play button.

[image]

It should build just fine.

[image]

Locate the kext in finder and proceed like I described in a post above.

chris1111, 2018-03-02:
Got it thanks 
[[image]

chris1111, 2018-03-02:
I have make mistake, my binaries not build correctly 🧐

DevAndArtist, 2018-03-02:
@chris1111 I only use `GPUSensors.kext` in /S/L/E, all other kext are from older 1440 release.

Have you set the Version and Build as shown in my screenshot? After reboot (don't forget to rebuild permissions and cache using kext-wizzard) open terminal and type `kextstat` and search for `GPUSensors`. If it's there then it's running correctly.

chris1111, 2018-03-02:
@DevAndArtist No I use it in Clover, I ever used FakeSMC in clover Why S/L/E ?

DevAndArtist, 2018-03-02:
It's up to you where you'd install them, both should work I guess. I think MultiBeast will install FakeSMC into /S/L/E by default so I just did that. You can always try /EFI/EFI/CLOVER/kexts/Other and see if it works.

chris1111, 2018-03-02:
My question is Why GeforceSensor.kext and FakeSMC 3.5.1  from Slice works with my Geforce GT 1030 Since long time and here I have trouble ?

DevAndArtist, 2018-03-02:
@chris1111 no idea sorry.

dmatora, 2018-03-02:
@chris1111 there are much more people involved into developing slice fork

chris1111, 2018-03-02:
Thank you anyway mr @DevAndArtist and  @dmatora  😇

dmatora, 2018-03-02:
>I have make mistake, my binaries not build correctly 🧐 system not bootable
I don't think you can build wrong kexts, they are either done building or not
system not bootable most definitely means you didn't install kexts properly

chris1111, 2018-03-02:
Kext is in Clover and the build looks good but system stall very shortly

DevAndArtist, 2018-03-02:
If it's `AppleACPICPU` thing then your `FakeSMC` is broken, just use the one from 6.26.1440 binary build.

chris1111, 2018-03-02:
For now I restart the system with a rescue SSD, thats not a problem for me Its hang at ACPI some thing like this very fast, I know its in relation of my build, Fakesmc must not have built correctly

DevAndArtist, 2018-03-02:
Replace your compiled FakeSMC with the one from here https://github.com/kozlek/HWSensors/releases/download/6.26/HWSensors.6.26.1440.Binaries.dmg.zip and that's it. 

I use all kext from this package *except* for the `GPUSensors` which is the one we compiled previously.

For me everything is in S/L/E, but in CLOVER it should also be fine.

If you added, deleted or edited any kext in /S/L/E then you MUST rebuild permissions and restore caches.

dmatora, 2018-03-02:
@chris1111 I've sent build 1462 to your email

chris1111, 2018-03-02:
@dmatora  I revive a Email and .dmg I test the file right now thank you

chris1111, 2018-03-02:
Its work !! 😎 Thank you mr @dmatora 
[image]

dmatora, 2018-03-02:
@chris1111 you're welcome

chris1111, 2018-03-02:
@dmatora do you have the same version of HWMonitor.app please

chris1111, 2018-03-02:
Ok got it I build successfully the same Build
 [image]

chris1111, 2018-03-19:
Hi @DevAndArtist  I resolved the compile issue  
my mistake is I forgot to put a bundle Identifier 
[[image]

and the other kext is 
org.hwsensors.driver.ACPISensors
org.hwsensors.driver.CPUSensors
org.hwsensors.driver.GPUSensors
org.hwsensors.driver.LPCSensors

all kext is compile great and its working like a charm, thanks both of you @DevAndArtist , @dmatora

emiliogonzalezjunior, 2018-03-27:
I was unable to boot with my build version of the kext's. Can anyone share their build last version of the kexts? 
Thanks in advance!

DevAndArtist, 2018-03-27:
@emiliogonzalezjunior have you rebuild your caches and restored permissions after messing around with /S/L/E (assuming your kexts are there)?

emiliogonzalezjunior, 2018-03-27:
I use the EFI/ other folder. I never installed anything in S/L/E

chris1111, 2018-03-27:
@Emilio  you can this Package, build  6.26. 1464

emiliogonzalezjunior, 2018-03-27:
Thank you very much @chris1111 !